wzab/agwb #8: Bus behavior on internal address over range.

m-kru, 2019-04-09:
Right now such scenarios in `*_wb.vhd` files are handled in the following way:

```
case int_addr is
...
...
...
when others =>
    int_regs_wb_m_i.dat <= x"A5A5A5A5";
    int_regs_wb_m_i.ack <= '1';
    int_regs_wb_m_i.err <= '0';
end case;
```

Is it correct? Wouldn't it be more appropriate to indicate an error on the bus?

wzab, 2019-05-25:
Yes, it would be, however when connecting to the AXI bus in certain SoCs it hangs the Linux.
That's why I have decided to return certain specific pattern.
It may be modified when the problem with hanging the system is resolved.

wzab, 2020-06-22:
I have modified it to return error.

closed 2020-06-22 by wzab

m-kru, 2020-06-23:
How did you resolve the Linux hang problem?