wzab/agwb #12: Defining default values for register fields has no effect.

m-kru, 2019-05-11:
Following xml definition has no effect in terms of default values.

```
<creg name="CONFIG">                                                                                                                                                                                        
     <field name="RX_WIDEBUS_MODE" width="1" default="0x0"/>                                                                                                                                                   
     <field name="INIT_VALUE" width="20" default="0x0"/>                                                                                                                                                        
</creg>
```

wzab, 2019-05-23:
It seems, that currently it works only on the register level, not on the bitfields level. So you may do

    <creg name="CONFIG" default="0x0">                                                                                                                                                                                        
       <field name="RX_WIDEBUS_MODE" width="1" />                                                                                                                                                   
       <field name="INIT_VALUE" width="20" />                                                                                                                                                        
    </creg>

but you can't define defaults for the bitfields. It must be checked if that can be easily modified.

wzab, 2019-05-24:
The default values for bitfields are supported after commit https://github.com/wzab/addr_gen_wb/commit/41c202387d7c9190b36c335a10249c90c4724419

closed 2019-05-24 by wzab