wzab/agwb #45: Generation of multiple AGWB tree instances to be used in the same design

wzab, 2020-12-20:
Current implementation of AGWB has problems with FPGAs hosting e.g. multiple PCIe devices (an example may be huge Xilinx FPGA with two SLRs).
In that case we may have two independent and slightly different WB-based devices, which must be implemented in the same device. The simplest solution would be to prepare two separate top blocks with two different sets of parameters, but that may result in generation of entities with the same names and e.g. different types of ports. 
This could be solved, by allowing the user to specify the name of the library generated for each top (e.g. instead of "agwb" in both cases, "agwb_slr1" for the first top, and "agwb_slr2" for the second top.
That solution however generates problems related to interfacing with the user logic, because the same entity can't include either one or the another library depending on generics (which could pass the SLR number).

wzab, 2020-12-30:
The proposed solution (being implemented now) is described in the [Wiki](../wikis/Multiple-AGWB-trees).

wfjm, 2025-08-17:
Isn't that solved with variants since a long time ?  
I guess, that this issue can be closed.