wzab/agwb #66: Explicit register addresses as constants in HDL package

lepommosaure, 2023-12-11:
Hi,
First of all, thank you for this generator! 
I think it would be really nice to have the generation process generate the register address mapping in VHDL as constants in a package, and not as magic values directly in the interconnect files. Is it something that has been done somewhere? I tried to look around with no avail, but some branch named "fixed_addresses" exists, so I'm asking before trying to reimplement anything :)

Thanks!

m-kru, 2024-01-08:
What is the point of such an approach? What is your use case? The point of the tool is to abstract away explicit address handling.

lepommosaure, 2024-01-08:
Hi,
My use case is to simplify direct (no CPU) VHDL requests issuing between different entities.
When you state that the explicit address handling is abstracted, I haven't found any abstraction for the VHDL part, only for C drivers, is that correct?

m-kru, 2024-01-08:
Ok, I see what you want to do. This is indeed currently not supported. I think generating package, for each block, containing constants with addresses and masks should be fairly easy. Unfortunately, I think no one has enough time to do it now. Maybe you can try to do it yourself?

wzab, 2024-01-12:
Hello, when communicating from one VHDL entity to another, the WB bus is typically not the most efficient channel. 

The branch [fixed_addresses](https://github.com/wzab/agwb/tree/fixed_addresses) was created to enable reusing the blocks created outside AGWB in the AGWB-generated HW and SW (please see the commit log messages in that branch).
Its development was suspended, as the need for that functionality in our main project has disappeared. 

Anyway, it is not something you need for your application. What you need, is a new backend routine targetting the VHDL output. Currently there are routines for [Python](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L1182), [Forth](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L1151), [CBM DCA XML](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L935), [IPbus XML](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L1000), [C headers](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L1062).
The routine [gen_vhdl](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L488) generates the implementation, not the addresses.

Please note, that the above routine may get redefined for child blocks (e.g. [Python for BlackBox](https://github.com/wzab/agwb/blob/b433c8a1f5b36137a518f238b077b71a8aa12e4a/src/wb_block.py#L1325)).

When proposing the new backend routine, the format of the generated representation of addresses should be decided first. Our experience is that once it is decided and implementating is started, it is difficult to redefine it. So, how the hierarchical set of addresses should be implemented in VHDL for hardware WB master?